danielholanda/LeFlow #2: Issue while running test_all.py

bagumondigi, 2018-08-14:
Hello Daniel,

Firstly, thank you for your great project, I'm very appreciate that.

As the title, I got an issue when I try to run the test_all.py . The process run well through first 3 modules: 03_vecmul_b_f , 04_dense_a, 06_softmax_a but failed at 09_con2d_a or 12_maxp_a ( --fast mode ). I can't figure out what cause this problem. Did you experienced these errors before  ? ( please checked attached file) . Thank you :) 

P/s: I tried to remove '09_conv2d_a','12_maxp_a' from test_all.py and the process finished successfully. But it's not a good solution, isn't it ? :) 
[image]

bagumondigi, 2018-08-14:
Hi Daniel,
1/ Yes, I used virtual machine for Legup 4.0 download from: http://legup.eecg.utoronto.ca/getstarted.php
2/ I used command `sudo pip install tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl --ignore-installed six` in directory src/tensorflow

danielholanda, 2019-01-10:
Hi bagumondigi,

The technique that we used to create in this tool could also very simply be used to do the same with Vivado HLS or any other HLS tool that is based on LLVM. The problem is that those tools are not open-source, so I can't really modify them in any way.