danielholanda/LeFlow #3: RUN TEST

pocandraspoc, 2018-08-17:
Hi,

I would like to run the tests one by one but somsing went side ways:

legup@legup-vm:~/Downloads/LeFlow-master/test/01_vecmul_a$ ~/Downloads/LeFlow-master/src/LeFlow 01_vecmul_a.py
INFO: Creating project...
INFO: Cleaning previous files...
INFO: Generating IR from tensorflow...
INFO: Cleaning unused dumped files...
Traceback (most recent call last):
  File "/home/legup/Downloads/LeFlow-master/src/LeFlow", line 195, in <module>
    run_leflow(args.file_name,tool_path)
  File "/home/legup/Downloads/LeFlow-master/src/LeFlow", line 79, in run_leflow
    shutil.copy(project_folder+"ir/ir-cluster_0__XlaCompiledKernel_true__XlaNumConstantArgs_0__XlaNumResourceArgs_0__module-with-opt.ll",project_folder+project_name+"_ir_1.ll")
  File "/usr/lib/python2.7/shutil.py", line 119, in copy
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 82, in copyfile
    with open(src, 'rb') as fsrc:
IOError: [Errno 2] No such file or directory: '01_vecmul_a_files/ir/ir-cluster_0__XlaCompiledKernel_true__XlaNumConstantArgs_0__XlaNumResourceArgs_0__module-with-opt.ll'

I am using the legup virtual box
 and did chmod +x Leflow.

If you can give me any suggestion where i messed up that would be great thanks!

pocandraspoc, 2018-08-17:
Sorry, was my fault I have downloaded the "same" versionof TF from conda cloud but re installed from your lib and it is fine now

closed 2018-08-17 by pocandraspoc

BrayanBarrios, 2019-03-05:
I have a similar issue after running `python test_all.py --fast`

I'm running the legup virtual box and I cloned your repo.

Running test for 03_vecmul_b_f
    Generating the circuit...
Traceback (most recent call last):
  File "../../src/LeFlow", line 200, in <module>
    run_leflow(args.file_name,tool_path)
  File "../../src/LeFlow", line 79, in run_leflow
    shutil.copy(project_folder+"ir/ir-cluster_0__XlaCompiledKernel_true__XlaNumConstantArgs_0__XlaNumResourceArgs_0__module-with-opt.ll",project_folder+project_name+"_ir_1.ll")
  File "/usr/lib/python2.7/shutil.py", line 119, in copy
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 82, in copyfile
    with open(src, 'rb') as fsrc:
IOError: [Errno 2] No such file or directory: '03_vecmul_b_f_files/ir/ir-cluster_0__XlaCompiledKernel_true__XlaNumConstantArgs_0__XlaNumResourceArgs_0__module-with-opt.ll'
    Generating new inputs and running Tensorflow with them...
    Testing circuit using Modelsim with new inputs...
cp: cannot stat ‘03_vecmul_b_f_files/tfArgs/param*.mif’: No such file or directory
/home/legup/legup-4.0/examples//../llvm/Release+Asserts/bin/llvm-dis: Could not open 03_vecmul_b_f.prelto.1.bc: No such file or directory

make[1]: *** [all] Error 1
make: *** [03_vecmul_b_f.v] Error 2
Traceback (most recent call last):
  File "test_all.py", line 92, in <module>
    modelsim_result = np.array(mif.getModelsimMem(folder+"_files/memory_dump.txt"))
  File "../src/processMif.py", line 77, in getModelsimMem
IOError: [Errno 2] No such file or directory: '03_vecmul_b_f_files/memory_dump.txt'

I don't know what I'm doing wrong. I hope you can help me, thank you!

BrayanBarrios, 2019-03-05:
I solved my problem installing an older version of the modified tensorflow.

https://github.com/danielholanda/LeFlow/issues/12#issuecomment-461877820

SAB95852, 2020-01-05:
python LeFlow tf1.py
INFO: Creating project...
INFO: Cleaning previous files...
INFO: Generating IR from tensorflow...
INFO: Cleaning unused dumped files...
Traceback (most recent call last):
  File "LeFlow", line 200, in <module>
    run_leflow(args.file_name,tool_path)
  File "LeFlow", line 79, in run_leflow
    shutil.copy(project_folder+"ir/ir-cluster_0__XlaCompiledKernel_true__XlaNumConstantArgs_0__XlaNumResourceArgs_0__module-with-opt.ll",project_folder+project_name+"_ir_1.ll")
  File "/usr/lib/python2.7/shutil.py", line 139, in copy
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 96, in copyfile
    with open(src, 'rb') as fsrc:
IOError: [Errno 2] No such file or directory: 'tf1_files/ir/ir-cluster_0__XlaCompiledKernel_true__XlaNumConstantArgs_0__XlaNumResourceArgs_0__module-with-opt.ll'

I run a simple tensorflow  array multiplication.The above error occured.Please suggest any solution