danielholanda/LeFlow #6: Synthesis issue in classificationMNIST

mahinlman, 2018-08-28:
Info: Running Quartus II 64-Bit Analysis & Synthesis
	Info: Version 15.0.0 Build 145 04/22/2015 SJ Web Edition
	Info: Processing started: Tue Aug 28 06:33:51 2018
Info: Command: quartus_map --read_settings_files=on --write_settings_files=off mnist -c mnist
Warning (20028): Parallel compilation is not licensed and has been disabled
Critical Warning (10191): Verilog HDL Compiler Directive warning at stratixv_pcie_hip_atoms_ncrypt.v(27): text macro "pragma" is undefined
Error (10170): Verilog HDL syntax error at stratixv_pcie_hip_atoms_ncrypt.v(28) near text "protect";  expecting ";"
Error (10170): Verilog HDL syntax error at stratixv_pcie_hip_atoms_ncrypt.v(30) near text "protect";  expecting ";"
Error (10170): Verilog HDL syntax error at stratixv_pcie_hip_atoms_ncrypt.v(33) near text "protect";  expecting ";"
Error (10170): Verilog HDL syntax error at stratixv_pcie_hip_atoms_ncrypt.v(33) near text ")";  expecting ";"
Error (10149): Verilog HDL Declaration error at stratixv_pcie_hip_atoms_ncrypt.v(37): identifier "line_length" is already declared in the present scope
Error (10149): Verilog HDL Declaration error at stratixv_pcie_hip_atoms_ncrypt.v(37): identifier "bytes" is already declared in the present scope
Error (10170): Verilog HDL syntax error at stratixv_pcie_hip_atoms_ncrypt.v(37) near text ")";  expecting ";"
Warning (10259): Verilog HDL error at stratixv_pcie_hip_atoms_ncrypt.v(21033): constant value overflow
Warning (10259): Verilog HDL error at stratixv_pcie_hip_atoms_ncrypt.v(41578): constant value overflow
Warning (10259): Verilog HDL error at stratixv_pcie_hip_atoms_ncrypt.v(51863): constant value overflow
Warning (10259): Verilog HDL error at stratixv_pcie_hip_atoms_ncrypt.v(98647): constant value overflow
Info (12021): Found 0 design units, including 0 entities, in source file stratixv_pcie_hip_atoms_ncrypt.v
Warning (12018): Entity "stratixv_hssi_gen3_pcie_hip" will be ignored because it conflicts with Quartus II primitive name
Info (12021): Found 1 design units, including 1 entities, in source file stratixv_pcie_hip_atoms.v
Critical Warning (10191): Verilog HDL Compiler Directive warning at stratixv_hssi_atoms_ncrypt.v(27): text macro "pragma" is undefined
Error (10149): Verilog HDL Declaration error at stratixv_hssi_atoms_ncrypt.v(28): identifier "begin_protected" is already declared in the present scope
Error (10170): Verilog HDL syntax error at stratixv_hssi_atoms_ncrypt.v(28) near text "protect";  expecting ";"
Error (10149): Verilog HDL Declaration error at stratixv_hssi_atoms_ncrypt.v(30): identifier "encrypt_agent_info" is already declared in the present scope
Error (10170): Verilog HDL syntax error at stratixv_hssi_atoms_ncrypt.v(30) near text "protect";  expecting ";"
Error (10149): Verilog HDL Declaration error at stratixv_hssi_atoms_ncrypt.v(32): identifier "key_keyname" is already declared in the present scope
Error (10149): Verilog HDL Declaration error at stratixv_hssi_atoms_ncrypt.v(33): identifier "key_method" is already declared in the present scope
Error (10170): Verilog HDL syntax error at stratixv_hssi_atoms_ncrypt.v(33) near text "protect";  expecting ";"
Error (10149): Verilog HDL Declaration error at stratixv_hssi_atoms_ncrypt.v(33): identifier "line_length" is already declared in the present scope
Error (10149): Verilog HDL Declaration error at stratixv_hssi_atoms_ncrypt.v(33): identifier "bytes" is already declared in the present scope
Error (10170): Verilog HDL syntax error at stratixv_hssi_atoms_ncrypt.v(33) near text ")";  expecting ";"
Error (10149): Verilog HDL Declaration error at stratixv_hssi_atoms_ncrypt.v(37): identifier "line_length" is already declared in the present scope
Error (10149): Verilog HDL Declaration error at stratixv_hssi_atoms_ncrypt.v(37): identifier "bytes" is already declared in the present scope
Error (10170): Verilog HDL syntax error at stratixv_hssi_atoms_ncrypt.v(37) near text ")";  expecting ";"
Warning (10259): Verilog HDL error at stratixv_hssi_atoms_ncrypt.v(17474): constant value overflow
Warning (10259): Verilog HDL error at stratixv_hssi_atoms_ncrypt.v(28564): constant value overflow
Warning (10259): Verilog HDL error at stratixv_hssi_atoms_ncrypt.v(76875): constant value overflow
Warning (10259): Verilog HDL error at stratixv_hssi_atoms_ncrypt.v(109657): constant value overflow
Warning (10259): Verilog HDL error at stratixv_hssi_atoms_ncrypt.v(138601): constant value overflow
Warning (10259): Verilog HDL error at stratixv_hssi_atoms_ncrypt.v(175293): constant value overflow
Info (12021): Found 0 design units, including 0 entities, in source file stratixv_hssi_atoms_ncrypt.v
Warning (12018): Entity "stratixv_channel_pll" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_pma_aux" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_avmm_interface" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_10g_rx_pcs" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_10g_tx_pcs" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_8g_pcs_aggregate" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_8g_rx_pcs" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_8g_tx_pcs" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_common_pcs_pma_interface" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_common_pld_pcs_interface" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_gen3_rx_pcs" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_gen3_tx_pcs" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_pipe_gen1_2" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_pipe_gen3" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_pma_cdr_att" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_pma_deser_att" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_pma_rx_att" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_pma_rx_buf" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_pma_rx_deser" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_pma_ser_att" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_pma_tx_att" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_pma_tx_buf" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_pma_tx_cgb" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_pma_tx_ser" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_rx_pcs_pma_interface" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_rx_pld_pcs_interface" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_tx_pcs_pma_interface" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_tx_pld_pcs_interface" will be ignored because it conflicts with Quartus II primitive name
Warning (12018): Entity "stratixv_hssi_refclk_divider" will be ignored because it conflicts with Quartus II primitive name
[... 1146 lines cut ...]
	Info (12023): Found entity 18: lpm_counter
	Info (12023): Found entity 19: lpm_latch
	Info (12023): Found entity 20: lpm_ff
	Info (12023): Found entity 21: lpm_shiftreg
	Info (12023): Found entity 22: lpm_ram_dq
	Info (12023): Found entity 23: lpm_ram_dp
	Info (12023): Found entity 24: lpm_ram_io
[... cut ...]
Info (144001): Generated suppressed messages file /home/legup/LeFlow/examples/classificationMNIST/classificationMNIST_files/output_files/mnist.map.smsg
Error: Quartus II 64-Bit Analysis & Synthesis was unsuccessful. 35 errors, 425 warnings
	Error: Peak virtual memory: 1325 megabytes
	Error: Processing ended: Tue Aug 28 06:34:10 2018
	Error: Elapsed time: 00:00:19
	Error: Total CPU time (on all processors): 00:00:16
Error (293001): Quartus II Flow was unsuccessful. 37 errors, 425 warnings

I'm using quartus provided by legup machine

danielholanda, 2018-08-29:
Hi Mahinlman,

To synthesize the design you first need to generate the hardware using LeFlow. I assuming that you are doing this right.
Then, cd into the generated folder and run "make p" to create a project and "make f" to synthesize it using Quartus. 
The board that we are normally targeting is a StratixIV DE4-530 (you can see this at src/LeFlow_config.tcl). Unfortunately you will need a Quartus license to synthesize a design for this board. If you don't have a license you can try to compile it for a CycloneV DE1-SoC (some designs might not fit/work with this board). To do this, simply change the src/LeFlow_config.tcl file to:

set_project CycloneV DE1-SoC Tiger_SDRAM
set_parameter INFERRED_RAMS 1

and generate your hardware again using LeFlow.

Let me know if that helps.

mahinlman, 2018-08-29:
Hi Danielholanda,

Thanks, I'm able to synthesize now.
Is there any option for using other boards (Intel Arria 10 GX FPGA) also.

danielholanda, 2018-08-29:
I guess this is more a LegUp question than a LeFlow question. 

LegUp natively support a couple of boards. You can check which ones at /home/legup/legup-4.0/boards. Theoretically you can also add support for another boards that are not there by providing information of the resources available in the FPGA you are targeting. I have never tried this, though. 

You can find some more information about that here:
http://legup.eecg.utoronto.ca/docs/4.0/xilinx.html

closed 2018-08-29 by danielholanda