danielholanda/LeFlow #10: Devices

jamesjjk, 2018-10-01:
@danielholanda Awesome project? Which FPGA's have you used to benchmark/test the example lib successfully on?

danielholanda, 2018-10-03:
Hi James,

Thanks for your feedback!

The examples should work on most of the boards supported by LegUp. You can find them in the folder "boards" inside of the LegUp folder. I'm now mostly using a DE1-SoC for my tests.

To change the target board change
set_project StratixIV DE4-530 Tiger_DDR2
to
set_project CycloneV DE1-SoC Tiger_SDRAM
in the LeFlow_config.tcl file

The interesting examples are maxpoolMNIST, convolutionLenna and classificationMNIST. The first one will fit very easily in the board. The second one will fit as long as you can fit the original image in the FPGA. Since each pixel is 32-bits by default (which we intend to change in the future) you will have to create smaller convolutions and process things over multiple batches. The classificationMNIST also fits nicely in a DE1-SoC and I'm currently working on a nice interactive demo to showcase it. Hopefully I will upload a video with it here soon.

jamesjjk, 2018-10-03:
@danielholanda Thanks for the quick reply I think this is a great project. 

Have you ever tested on a Xilinx device or even a Xilinx MPSOC device? 
I know legup can output generic verilog (but without floatingpoint support).

Xilinx has its own C Synthesis tool https://www.xilinx.com/products/design-tools/software-zone/sdsoc.html

Would be great to find a solution from LLVM-IR to Xilinx. Thanks for your thoughts.

danielholanda, 2018-10-04:
Hi,

We never tested things with Xilinx boards, but it should work as long as you can slightly modify the LegUp files for it. As you said, LegUp can output generic Verilog. 

It would be actually very interesting to do the same with Vivado HLS, but this would require access to their source code (or at least a way of inserting IR instead of C), which we currently don't have.

closed 2018-10-15 by danielholanda