danielholanda/LeFlow #12: "No such file or directory 03_vecmul_b_f_files" during running test_all.py

kp2377, 2019-01-04:
When I tried to run test all after installation, I am getting following error:-

Traceback (most recent call last):
  File "../../src/LeFlow", line 200, in <module>
    run_leflow(args.file_name,tool_path)
  File "../../src/LeFlow", line 79, in run_leflow
    shutil.copy(project_folder+"ir/ir-cluster_0__XlaCompiledKernel_true__XlaNumConstantArgs_0__XlaNumResourceArgs_0__module-with-opt.ll",project_folder+project_name+"_ir_1.ll")
  File "/usr/lib/python2.7/shutil.py", line 119, in copy
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 82, in copyfile
    with open(src, 'rb') as fsrc:
IOError: [Errno 2] No such file or directory: '03_vecmul_b_f_files/ir/ir-cluster_0__XlaCompiledKernel_true__XlaNumConstantArgs_0__XlaNumResourceArgs_0__module-with-opt.ll'
	Generating new inputs and running Tensorflow with them...
	Testing circuit using Modelsim with new inputs...
/home/legup/legup-4.0/examples//../llvm/Release+Asserts/bin/llvm-dis: Could not open 03_vecmul_b_f.prelto.1.bc: No such file or directory

make[1]: *** [all] Error 1
make: *** [03_vecmul_b_f.v] Error 2
Traceback (most recent call last):
  File "test_all.py", line 92, in <module>
    modelsim_result = np.array(mif.getModelsimMem(folder+"_files/memory_dump.txt"))
  File "../src/processMif.py", line 77, in getModelsimMem
IOError: [Errno 2] No such file or directory: '03_vecmul_b_f_files/memory_dump.txt'

danielholanda, 2019-01-04:
Hi,

Can you confirmed that you installed our modified version of Tensorflow and that you are using the LegUp VM? If not, what is your setup?

Thanks

danielholanda, 2019-01-08:
Hi,

By typing "pip install tensorflow" you are installing the original version of Tensorflow, not our modified version. Installing our version of Tensorflow should solve the problem. 

I know that if pip is not updated/upgraded some errors might occur when installing a whl file, but it should be pretty straight forward to find the solution for problems like those in the web. Usually, all you need is a "pip install --upgrade" or something similar.

closed 2019-01-08 by danielholanda

kp2377, 2019-01-10:
Thank you so much for the help. I did worked.

Riyuzakii, 2019-02-08:
@danielholanda 
I'm using the legup-VM and I installed your version of tensorflow as well.
[image]
any suggestion on how to fix this?

danielholanda, 2019-02-08:
Hi,

It is very weird that this is happening during the first test. We recently updated the tensorflow whl file. Could you verify that this problem is also happening with the old version? 

https://github.com/danielholanda/LeFlow/tree/7504ea9a5fcde898d6962eca5efbc5602bb522f2/src/tensorflow

Riyuzakii, 2019-02-09:
@danielholanda I tried with the older version, still the same error.

danielholanda, 2019-02-11:
Unfortunately I was not able to reproduce your error. The only thing I can think of is that you didn't install our version of tensorflow correctly. You can check thin by trying to run one of the examples without LeFlow (python name_of_script.py) and then run it with LeFlow (../../src/LeFlow name_of_script.py). Some of the examples (not tests) also require matplotlib. Make sure to install it before using LeFlow.

SAB95852, 2020-01-04:
python test_all.py

Running test for 01_vecmul_a
	Generating the circuit...
	Generating new inputs and running Tensorflow with them...
	Testing circuit using Modelsim with new inputs...
make: *** No rule to make target 'v'.  Stop.
Traceback (most recent call last):
  File "test_all.py", line 92, in <module>
    modelsim_result = np.array(mif.getModelsimMem(folder+"_files/memory_dump.txt"))
  File "../src/processMif.py", line 77, in getModelsimMem
IOError: [Errno 2] No such file or directory: '01_vecmul_a_files/memory_dump.txt'

SAB95852, 2020-01-04:
@danielholanda 
python test_all.py

Running test for 01_vecmul_a
Generating the circuit...
Generating new inputs and running Tensorflow with them...
Testing circuit using Modelsim with new inputs...
make: *** No rule to make target 'v'. Stop.
Traceback (most recent call last):
File "test_all.py", line 92, in
modelsim_result = np.array(mif.getModelsimMem(folder+"_files/memory_dump.txt"))
File "../src/processMif.py", line 77, in getModelsimMem
IOError: [Errno 2] No such file or directory: '01_vecmul_a_files/memory_dump.txt'

danielholanda, 2020-01-08:
Hi,

You definitely need LegUp to make this work. The easiest way is to download their VM, but it is possible to install LegUp manually and make it work with LeFlow (but not recommended).

SAB95852, 2020-01-08:
Shall I have to download .vvd image file or just source code of legup 4.0

mingyi136, 2020-04-17:
Hai @danielholanda , nice work !

Is that workable if I use Tensorflow 1.15 to be compiled from scratch instead of using Tensorflow 1.6.0? Currently, I'm trying to convert Faster R-CNN model into verilog but seem like the model is not compatible with Tensorflow 1.6.0. (ValueError: No op named FusedBatchNormV3 in defined operations.) By the way, is that feasible to convert Faster R-CNN model into verilog using LeFlow?

Thanks in advance!

danielholanda, 2020-04-28:
Hi,

Not all operations of TensorFlow are supported by LeFlow. This is definitely going to be a bottleneck in your case. I would recommend sticking to the version of TensorFlow we currently support and adapting your model to it. 

Also remember that all weights/biases/activations must fit on-chip, which might be a limitation if you are trying to implement large networks like Faster R-CNN.