danielholanda/LeFlow #13: Fail to generate IR for reduce_mean and reduce_max

santacml, 2019-01-10:
Hello,

This tool is awesome! Unfortunately, I am running into a bit of trouble.

My network includes tf.reduce_mean() and tf.reduce_max() operations. No specific errors are given, but from what I understand looking through LeFlow, the IR fails to generate for these operations. It manifests by failing to copy the final output IR file. 

I can replace reduce_max() with some fancy reshaping and a tf.nn.max_pool(), but I do not see a way to replace the reduce_mean(). 

Do you know what the error could be? I'm happy to help contribute to the source if you are too busy and these functions are not implemented, but I'm not sure where to start.

Thanks!

Michael Santacroce

danielholanda, 2019-01-10:
Hi Michael,

Thanks for your message.

I never tried to use tf.reduce_mean() with LeFlow, but I think I might know what is causing the problem. 

An operation only works in LeFlow if the following conditions are true
(1) XLA code is structured in a simple way (see examples folder)
(2) The operation is implemented in XLA
(3) The IR generated by XLA does not contain any instructions not supported by LegUp (e.g. vectorized instructions)

If (1) and (2) are true, I think you should be able to generate IR from Tensorflow.

If (1) and (2) are true and (3) is not, you will get some errors when generating the circuit using LegUp. Sometimes, XLA has multiple ways of mapping Tensorflow operations to IR instructions (kernels). Our modified version of Tensorflow takes advantage of this and forces Tensorflow to always select kernels that work well with LegUp.

I assume that your problem might be caused by either the way in which your code is structured or by XLA not supporting this operation, but I can't be sure until I have a look at the code. Could you maybe send me a very short example that uses tf.reduce_mean()? Hopefully something that looks like the code I posted here https://github.com/danielholanda/LeFlow/issues/9

After that i should be able to give you further directions.

Thanks,

Daniel

santacml, 2019-01-10:
Thanks for the quick response!

Below is a code sample that uses reduce_mean based on that sample. I have confirmed that running the script with Python outputs some numbers, and LeFlow fails. Also, removing reduce_mean results in successful compilation. 

In my code I use the kwargs "axis" and "keepdims", which were apparently in flux around the version of the custom tensorflow you have. They existed, but existed alongside some kwargs that were going to be deprecated. I tried the other versions and they did not work with LegUp either.

Interestingly, when I run this example, it fails in a different way than my actual net. This time, the ir seems to be generated, but it simply says that LegUp failed to generate verilog. running the makefile manually results in this:

[image]

So maybe the issue is something with this Zeroinitializer? I'm not sure. Let me know if you need any additional info. Here is the code:

```python
import tensorflow as tf
import numpy as np
import sys
sys.path.append('../../src')

import processMif as mif

# This creates a random 32-bit floating point input for your variables
def rand(dim):
    return np.random.random(dim).astype(np.float32)

X = tf.placeholder(tf.float32, [1, 28,28,1])
test_image=rand([1,28,28,1])

conv1_w = tf.Variable(rand([5,5,1,8]))
conv1_b = tf.Variable(rand([8])) 

conv2_w = tf.Variable(rand([5,5,8,16])) 
conv2_b = tf.Variable(rand([16]))

dense1_w = tf.Variable(rand([16, 20])) 
dense1_b = tf.Variable(rand([20]))

dense2_w = tf.Variable(rand([20, 10])) 
dense2_b = tf.Variable(rand([10]))

with tf.Session() as sess:

    # Generating circuit
    with tf.device("device:XLA_CPU:0"):

        # Convolution 1
        conv1 = tf.nn.relu(tf.add(tf.nn.conv2d(X, conv1_w, strides=[1, 1, 1, 1], padding='SAME'),conv1_b))

        # Maxpool 1 
        maxp1=tf.layers.max_pooling2d(inputs=conv1, pool_size=[2, 2], strides=2)

        # Convolution 2
        conv2 = tf.add(tf.nn.conv2d(maxp1, conv2_w, strides=[1, 1, 1, 1], padding='SAME'),conv2_b)

        # Maxpool 2
        maxp2=tf.layers.max_pooling2d(inputs=conv2, pool_size=[2, 2], strides=2)

        # Reshaping array
        maxp2_reshaped = tf.nn.relu(tf.reshape(maxp2, shape=[1, 49, 16]))
        
        # using reduce_mean
        maxp2_reduced = tf.reduce_mean(maxp2_reshaped, axis=1, keepdims=False)
        
        # this can also be changed to reduce_max having the same effect on the output shape
        # maxp2_reduced = tf.reduce_max(maxp2_reshaped, axis=1, keepdims=False)
        
        # Dense layer 1 
        dense1 = tf.add(tf.matmul(maxp2_reduced, dense1_w)[0], dense1_b)

        # Dense layer 2
        dense1_reshaped = tf.reshape(dense1, shape=[1, 20])
        y = tf.add(tf.matmul(dense1_reshaped, dense2_w)[0], dense2_b)

    sess.run(tf.global_variables_initializer())
    result = sess.run(y,{X: test_image})

    # Print expected result
    print("Result Calculated: "+ str(result))

    # Creating .mif files to initialize the memories
    # To do this, go to <your_file>_ir_7.ll and check the list of params (inputs) and their sizes 
    param8 = dense2_b.eval();
    param7 = dense2_w.eval();
    param6 = dense1_b.eval();
    param5 = dense1_w.eval();
    param4 = conv2_b.eval();
    param3 = conv2_w.eval();
    param2 = conv1_b.eval();
    param1 = test_image;
    param0 = conv1_w.eval();
    mif.createMem([param0,param1,param2,param3,param4,param5,param6,param7,param8])
```

danielholanda, 2019-01-10:
Hi,

So apparently XLA is generating vectorized instructions for when it maps reduce_mean() to IR. You can see that by observing YOUR_CIRCUIT_ir_1.ll, which is the file generated by XLA. In this file you will find some operations such as "store <8 x float> ...", which I believe are not supported by LegUp. 

The way I solved a similar problem in the past was to dig into Tensorflow's source code and make it generate instructions using a kernel that does not emit vectorized instructions. Hopefully this does't look that hard for the reduce operation.

[This file ](https://github.com/danielholanda/LeFlow/blob/master/src/tensorflow/tensorflow_src/ir_emitter.cc) is the file used by XLA to generate IR. Inside of it there is a function called "EmitVectorizedReduce", which really sounds like the source of our problem. 

Look for "IrEmitter::HandleReduce(HloInstruction* reduce)" in this file. Apparently setting the if statement " if (!options::VectorizedReduceDisabled(hlo_module_config_))" to false should force XLA to emit plain llvm IR. You can do this by either forcing it to be false or you can find where this hlo_module_config_ option is set and change it. If you need to recompile Tensorflow, have a look at[ these instructions](https://github.com/danielholanda/LeFlow/tree/master/src/tensorflow).

Hopefully this works. If it does let me know. I would be happy to add those changes to LeFlow.

santacml, 2019-01-16:
Hello,

With a bit of trouble, I was able to get reduce_mean and reduce_max to work. However, it's still not working with my main network, which I mention at the end. Here is the fix:

The trick was that these functions use llvm_loop.cc, which is where vectorization needed to be disabled, in addition to the spot you pointed out.

To keep it simple, I changed one if statement:

```C++
    //if (prevent_vectorization_) {
    // ALWAYS prevent vectorization to allow LegUp to work with loops 
    // multiple paths lead here - this allows the least amount of code changes
    if (true) {
    result.push_back(llvm::MDNode::get(
        *ctx, {llvm::MDString::get(*ctx, kLlvmLoopVectorizeMDName),
               llvm::ConstantAsMetadata::get(ir_builder->getFalse())}));
  }
```

Normally, loops can be specified to be vectorized or not. This change will make it so that loops are always vectorized, regardless of what is given. Since this tensorflow is (hopefully) only ever going to be used in tandem with these tools, this should be fine. Additionally, because loops are now vectorized, it may be possible that more than just reduce_ops can be used. I previously tried using a recurrent network and had troubles, and I imagine the cause is the loop. I will test and update you if you like.

I have made a branch and will submit a pull request right now. I included an updated tensorflow compile, but I wouldn't be offended if you recompiled it yourself... I had a lot of difficulties with Bazel and wouldn't be surprised if I messed something up.

One issue, though. the test file I sent you now works, but the network I am trying to use this tool with is still failing when using reduce_mean or reduce_max. It fails like this:

[image]

I can't share the network source code and like I said, the example I wrote is working. The network works if I call the file with python normally, so the network itself is working. Do you have any idea what the issue is?

danielholanda, 2019-01-22:
Hi,

This is amazing! Glad you were able to figure it out. This is a good contribution to the code!

I was on a conference last week and still didn't manage to find time to review your pool request, but will do it soon (hopefully over the weekend) and will let you know.

santacml, 2019-01-22:
I'm very glad I could help! This is all fascinating to me. No worries, I hope you enjoyed the conference.

With regards to the problem, I've been reading a lot about how tensorflow works in the backend, and I somewhat understand a separate vectorization issue. If you want to split this into a separate thread that would be fine with me but I'll dump here.

Could you confirm for me that I am understanding the following correctly:

The primary issue with recurrent nets and XLA is that they use while_loops (due to an unknown number of timesteps). Because while_loops depend on the while_loop conditional, the symbolic graph splits into separate "clusters", resulting in dead nodes and therefore multiple IR outputs from XLA. This likewise affects tf.nn.embedding_lookup, as the lookup uses a boolean mask as a conditional. I got all that from [here](https://www.playingwithpointers.com/blog/control-flow-and-tf-xla.html), [here](https://groups.google.com/forum/#!topic/xla-dev/Z0lpN69fkTQ), [here](https://groups.google.com/forum/#!topic/xla-dev/cgMgzdjlOQI), and [here](https://groups.google.com/forum/#!topic/xla-dev/IqLyL67cemI).

This is an issue for me as my network also uses and tf.nn.embedding_lookup. It wraps all the way back to this thread - I replaced embedding_lookup with tf.gather() as the former is a generalization of the latter to try to avoid the conditional. It looks like it is avoiding the conditional successfully, however, I believe tf.gather() is experience the same problem as reduce_mean in that it's generating vectorized LLVM. I am struggling to find where that occurs and if it's possible to stop it from being vectorized. 

Let me know if that all makes sense, and like I said, I'm happy to split this into a separate issue, but I think it's the same type of deal that the vectorization isn't working. I can also generate a test example to demonstrate the issue.

danielholanda, 2019-01-23:
I actually never tried to implement recurrent networks with XLA, but I can see how this could be problematic. 

Let's start by looking at tf.gather(). Could you send me a small example showing the problem you are having?

santacml, 2019-01-23:
Haha, I think I may have gone on too much of a tangent there, if we can get tf.gather() working then that's all I need.

Here is the error on running a file with gather: 
[image]

The code that generated that error is below:

```Python
import tensorflow as tf
import numpy as np
import sys
sys.path.append('../LeFlow/src')

import processMif as mif

# This creates a random 32-bit floating point input for your variables
def rand(dim):
    return np.random.random(dim).astype(np.float32)

X = tf.placeholder(tf.int32, [1, 28])
test_indeces=(rand([1,28])*10).astype(np.int32)

gather_weights = tf.Variable(rand([10,2]))

dense1_w = tf.Variable(rand([56, 5])) 
dense1_b = tf.Variable(rand([5]))

with tf.Session() as sess:

    # Generating circuit
    with tf.device("device:XLA_CPU:0"):
	
	gather = tf.gather(gather_weights, X)
	
	reshaped = tf.reshape(gather, (1, 56))

        dense1 = tf.add(tf.matmul(reshaped, dense1_w)[0], dense1_b)

        y = dense1

    sess.run(tf.global_variables_initializer())
    result = sess.run(y,{X: test_indeces})

    # Print expected result
    print("Result Calculated: "+ str(result))

    # Creating .mif files to initialize the memories
    # To do this, go to <your_file>_ir_7.ll and check the list of params (inputs) and their sizes
    param0 = gather_weights.eval()
    param6 = dense1_b.eval();
    param5 = dense1_w.eval();
    mif.createMem([param0, param5,param6])

```

Thank you very much for taking a look. If you can point me in the right direction again it would help greatly. I'm currently using this tool to deploy a network that I created for my Master's thesis, but I do have a working prototype, so if gather ends up not working it is not a huge deal.

santacml, 2019-01-29:
Thanks for your continued help with the issues I'm having.

I was able to split the load/getelemptr instructions in llvmSyntaxDowngrade.py, but it's having a different issue now. Fomr _ir_4 to _ir_5, it's having trouble handling the %params variable I think?

Here is a function in _ir_4:

[image]

And the function in _ir_5:

[image]

From my understanding, restructureMainSignature.py should be handling this? I'm not quite sure what's going on there, do you have any advice?

Thanks,

Michael Santacroce

danielholanda, 2019-02-04:
Hi,

I think I have an idea of what might be causing this problem. restructureMainSignature always assumes that each "param" is an array. When this is the case, the sequence of IR operations performed to use each of those params is very characteristic (gep [optional], load, gep [optional], bitcast). LeFlow transforms all of those operations to a gep of the global param followed by a load. 

In this case that you showed, it looks like the "param" is a single variable (not an array), which is something that I never encountered before.  restructureMainSignature (more specifically promoteToGlobal()) would need to be changed to also understand what is going on in this situation. 

Also, I checked your pull request and everything looks good! Thanks for your contribution!

santacml, 2019-02-07:
Thanks for the tips! I am glad that the pull request helped out!

I was able to edit restructureMainSignature to handle the LLVM (at least, I think it's doing the right thing), however, there are still a lot of issues. My impression from snooping around is that a lot of work would need to be done in the Tensorflow XLA compilation backend to support gather.

Unfortunately, this semester is actually my last semester at school for my Master's, so I'm kind of in the middle of writing a thesis / doing coursework / etc. I am able to use LeFlow without tf.gather to build a network with a performance hit, so I will likely stick with that as a proof-of-concept. I'd love to look deeper into Tensorflow as this is some really cool stuff but it would have to be after I graduate. I do have some questions for you so look out for an email from me (I'll email holand.daniel@gmail.com from your profile).

Thanks for all your help!

closed 2019-02-07 by santacml