danielholanda/LeFlow #15: See the schematic

roSimoes, 2019-02-21:
Hello,
First of all, congratulations for the project.

Can you tell me the best way to see the schematic produced by LeFlow?
I tried to import the files .v into Vivado of your example 06, but the [Synth 8-27] primitive not supported [stratixiv_atom.v] erro appeared.
Is there any better way to view schematics?

danielholanda, 2019-02-21:
Hi,

Thanks. Glad you like the project!

The open-source version of the HLS tool we use only targets Intel boards by default. This means that you should use Quartus instead of Vivado. 

The easiest way to see the architecture in my opinion is to 
1 - Create a project using LeFlow (start with one of our simple examples)
2 - cd into the generated folder and type "make p" to create a Quartus project
3 -  Open the project on quartus and run Analysis and Synthesis
3 - Go to Tools -> Netlist Viewers -> RTL Viewer

Hope this helps.

roSimoes, 2019-02-22:
Hi, 

Is it possible to view the schematic in Quartus Prime Lite edition?
I can not open the project,  (you must have device support installed) error.

danielholanda, 2019-02-22:
Hi,

Quartus Lite edition only supports a couple boards and I assume that the Stratix IV is not one of them. You can change src/LeFlow_config.tcl to use another board that is supported by both LegUp and Quartus Lite edition. I assume that the Cyclone V should be supported. Try updating this file using the options below and generate the circuit again. 

`set_project CycloneV DE1-SoC Tiger_SDRAM
#set_project StratixIV DE4-530 Tiger_DDR2
set_parameter INFERRED_RAMS 1`

roSimoes, 2019-02-25:
I would like to thank you for your help, it has been very useful.
I have not yet been able to see the schematic, but I think it must be a problem with my version of Quartus.
If it is possible to give me some clue about the subject what happens is:

When I do "make p" LeFlow creates project but when I try to open the "top" QPF file I get the error
"" You must have device support installed befor you can open
to project. To download device support, go to the
  Download Center section on the "Intel FPGA" website.

Sorry for the inconvenience should be something easy but I can not solve it, my knowledge on this subject is reduced I am learning now

danielholanda, 2019-02-26:
Hi, 

According to their website Quartus Lite is supposed to support the DE1SoC.

http://fpgasoftware.intel.com/devices/ (look for cyclone V)
http://fpgasoftware.intel.com/18.1/?edition=lite

The only thing I can think of is that you might not be generating the circuit again with the new settings (running LeFlow again).

roSimoes, 2019-02-28:
Thank you.
Now I can see the schematic

closed 2019-02-28 by roSimoes