danielholanda/LeFlow #16: Raspi + Spartan 6

dadrake3, 2019-02-21:
I am looking to implement an edge ML inferencing accelerator for the raspi. I am wondering if it is possible to use leflow to compile a TensorFlow graph to run on the spartan 6 FPGA. I am not sure where to start and I am wondering if this would be possible within leflow

danielholanda, 2019-02-22:
Hi,

LeFlow uses the open version of LegUp HLS, which only supports Intel boards. In theory you could modify LegUp to support Xilinx boards, but I would only recommend you to do that if you have some good experience with FPGAs/HLS tools. Otherwise, If you want to learn more about ML by implementing the network yourself, but don't want to deal with RTL, you could try using Vivado HLS.

dadrake3, 2019-02-22:
Hi thanks, Ill just use an intel board in that case. I have FPGA experience but that's not what Im trying to spend my time on for this project. Are there any hardware requirements for the board itself? Im thinking of using the DE0 cyclone board by altera, would this be sufficient for running CNNs?

Also I have not found much about i/o interfacing with the board once the model has been compiled. Is it possible to chose specific communication protocols for I/O, I was hoping to communicate via the pis GPIOs with i2c or uart or something else along those lines. Would this be best for low latency communication or are there better options such as usb. 

Thanks again

danielholanda, 2019-02-22:
Hi,

I think it is a good idea to download the LegUp VM and have a look at the "boards" folder, which has a folder for each board they support. I normally use the DE1-SoC, which is big enough for some small CNNs, such as the one that I used to classify drawings of different fruits using Google's 'Quick, Draw!" dataset. Keep in mind that very gib designs will not fit, because LeFlow requires all weights to be stored on-chip.

I/O interfacing would be completely up to you. LeFlow can generate a network that will read inputs from an on-chip memory and write the results into another on-chip memory. In this case, you would have to manually change the RTL to first write what you want into the input memory, then run the network and later retrieve the outputs from the output memory.

closed 2019-02-28 by danielholanda