danielholanda/LeFlow #18: Unable to install tensorflow for LeFlow

kratos6753, 2019-03-15:
I have downloaded virtual image from http://legup.eecg.utoronto.ca/legup-4.0-ubuntu-64-14.04.vdi. And booted into the VM. Tried installing tensorflow by following the instructions in the README file. Below is the error message I got while installing.

legup@legup-vm:~$ sudo python -m pip install --upgrade pip
Cannot fetch index base URL https://pypi.python.org/simple/
Could not find any downloads that satisfy the requirement pip in /usr/lib/python2.7/dist-packages
Downloading/unpacking pip
Cleaning up...
No distributions at all found for pip in /usr/lib/python2.7/dist-packages
Storing debug log for failure in /home/legup/.pip/pip.log
legup@legup-vm:~$ sudo pip install tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl --ignore-installed six
Requirement 'tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl' looks like a filename, but the file does not exist
tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl is not a supported wheel on this platform.
Storing debug log for failure in /home/legup/.pip/pip.log

The pip.log file contains

/usr/bin/pip run on Fri Mar 15 05:13:15 2019
Requirement 'tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl' looks like a filename, but the file does not exist
tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl is not a supported wheel on this platform.
Exception information:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 257, in run
    InstallRequirement.from_line(name, None))
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 168, in from_line
    raise UnsupportedWheel("%s is not a supported wheel on this platform." % wheel.filename)
UnsupportedWheel: tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl is not a supported wheel on this platform.

I tried installing tensorflow from source by replacing the modified files given in the repository, but failed to do so.
Please help me solve the issue asap. Thanks in advance.

danielholanda, 2019-03-15:
Hi,

I heard some people had similar problems in the recent past. I think this
is related to the new whl file we uploaded. Could you try the old one and
tell me if it works?

https://github.com/danielholanda/LeFlow/blob/7504ea9a5fcde898d6962eca5efbc5602bb522f2/src/tensorflow/tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl

Thanks,

Daniel

kratos6753, 2019-03-16:
Thank you for your help daniel. I resolved the issue. It happens that I am unable to install the tensorflow via proxy server in my school, as it has some dependencies that need to be downloaded from the internet. I am closing the issue.

closed 2019-03-16 by kratos6753