danielholanda/LeFlow #21: where shoud I give the input?

hsiehchungting, 2019-04-18:
Hi daniel!
This is a great project. However, I have some questions for asking your help.

1. If i want to test a image for example input a number image to see what number is in the image , where can i set input? Which files or where should i revise?
2. I want to imigrate yoloV3 or tiny yolo to FPGA , can you give me some advices how to do it?

Thanks for your time! Hopefully to hear your response soon!

danielholanda, 2019-04-18:
Hi,

Thank you for your comments
1 - LegUp HLS maps all inputs and outputs to on-chip memories. Normally to insert values to those inputs you create a memory initialization file (mif) as shown here: https://github.com/danielholanda/LeFlow/issues/13 (look for createMem)
You can also manually edit the RTL to insert some inputs as I discussed here: https://github.com/danielholanda/LeFlow/issues/20

2 - This example here can be a good start: https://github.com/danielholanda/LeFlow/issues/13
Just remember that since all inputs and outputs are mapped into on-chip memories by legup you will not be able to fit large networks in the FPGA since it is easy to run out of on-chip memory.

hsiehchungting, 2019-04-27:
Hi daniel!
I don't know.
Why I repeat use ''../../src/LeFlow classificationMNIST.py'' to generate hardware.But I can find out top.qpf?
Can you tell me how to find that?
Thanks for your time! Hopefully to hear your response soon!

danielholanda, 2019-04-29:
Hi,

Just cd into the generated directory and run "make p". This will generate the Quartus project.

closed 2019-05-08 by danielholanda