danielholanda/LeFlow #22: llvm: unexpected comma

jk78346, 2019-06-24:
When I run the command: 
```
python test_all.py --fast
```
It seems like llvm Ir has some problem , here is the terminal output:
```
Running test for 03_vecmul_b_f
	Generating the circuit...
llvm-as: 03_vecmul_b_f_files/03_vecmul_b_f_ir_2.ll:8:36: error: expected comma after getelementptr's type
  %0 = getelementptr inbounds i8** %params, i64 1
                                   ^
opt: Unknown command line argument '-no-aa'.  Try: 'opt -help'
opt: Did you mean '-cfl-aa'?
opt: Unknown command line argument '-notti'.  Try: 'opt -help'
opt: Did you mean '-tti'?
opt: Unknown command line argument '-no-aa'.  Try: 'opt -help'
opt: Did you mean '-cfl-aa'?
opt: Unknown command line argument '-inline-cost'.  Try: 'opt -help'
opt: Did you mean '-inline'?
llvm-dis: Could not open 03_vecmul_b_f_files/03_vecmul_b_f_ir_3.bc: No such file or directory

llvm-as: 03_vecmul_b_f_files/03_vecmul_b_f_ir_3.ll: error: Could not open input file: No such file or directory
opt: 03_vecmul_b_f_files/03_vecmul_b_f_ir_3.bc: error: Could not open input file: No such file or directory
llvm-dis: Could not open 03_vecmul_b_f_files/03_vecmul_b_f_ir_4.bc: No such file or directory

Traceback (most recent call last):
  File "/home/kc.hsu/LeFlow/src/restructureMainSignature.py", line 189, in <module>
    ir=misc.readIR(input_file)
  File "/home/kc.hsu/LeFlow/src/misc.py", line 111, in readIR
    with open(input_file,'r') as f_in:
IOError: [Errno 2] No such file or directory: '03_vecmul_b_f_files/03_vecmul_b_f_ir_4.ll'
Traceback (most recent call last):
  File "/home/kc.hsu/LeFlow/src/restructureOperations.py", line 40, in <module>
    ir=misc.readIR(input_file)
  File "/home/kc.hsu/LeFlow/src/misc.py", line 111, in readIR
    with open(input_file,'r') as f_in:
IOError: [Errno 2] No such file or directory: '03_vecmul_b_f_files/03_vecmul_b_f_ir_5.ll'
Traceback (most recent call last):
  File "/home/kc.hsu/LeFlow/src/memBank.py", line 157, in <module>
    ir=misc.readIR(input_file)
  File "/home/kc.hsu/LeFlow/src/misc.py", line 111, in readIR
    with open(input_file,'r') as f_in:
IOError: [Errno 2] No such file or directory: '03_vecmul_b_f_files/03_vecmul_b_f_ir_6.ll'
llvm-as: 03_vecmul_b_f_files/03_vecmul_b_f_ir_7.ll: error: Could not open input file: No such file or directory
Traceback (most recent call last):
  File "../../src/LeFlow", line 203, in <module>
    run_leflow(args.file_name,tool_path)
  File "../../src/LeFlow", line 117, in run_leflow
    shutil.move(project_folder+project_name+"_ir_7.bc", project_folder+project_name+".prelto.1.bc")
  File "/usr/lib/python2.7/shutil.py", line 302, in move
    copy2(src, real_dst)
  File "/usr/lib/python2.7/shutil.py", line 130, in copy2
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 82, in copyfile
    with open(src, 'rb') as fsrc:
IOError: [Errno 2] No such file or directory: '03_vecmul_b_f_files/03_vecmul_b_f_ir_7.bc'
	Generating new inputs and running Tensorflow with them...
	Testing circuit using Modelsim with new inputs...

Running test for 04_dense_a
Traceback (most recent call last):
  File "test_all.py", line 52, in <module>
    os.chdir("./{}/".format(folder))
OSError: [Errno 2] No such file or directory: './04_dense_a/'
```

jk78346, 2019-06-24:
My tensorflow version is 1.6.0, while I'm not sure if successfully installed the modified version (3 files modified) or just original version. I seems no way to check that. When I tried to install the tensorfow-1.6.0-cp27-cp27mu-linux_x86_64.whl , it shows: 
```
ERROR: tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl is not a supported wheel on this platform.
```
and my 
```
uname -a 
```
is 4.13.0-36-generic x86_64

danielholanda, 2019-06-24:
Hi Kuan-Chieh,

I'm not sure, but it looks like there is a problem with the llvm version you are using. Are you using the LegUp VM and the Tensorflow whl file provided? Or do you have a different setup?

jk78346, 2019-06-24:
> Hi Kuan-Chieh,
> 
> I'm not sure, but it looks like there is a problem with the llvm version you are using. Are you using the LegUp VM and the Tensorflow whl file provided? Or do you have a different setup?

Hi, thanks for quick reply.
I know Legup VM is strongly recommended. However I'm using server already, screen response and performance is my big concern. I followed all the instructions under this link: http://legup.eecg.utoronto.ca/docs/4.0/gettingstarted.html
 Have you ever try no-VM ?

danielholanda, 2019-06-24:
Hi Kuan-Chieh,

I'm not sure, but it looks like LeFlow is trying to use a different version from LLVM than what it is supposed to. You can check this by running the command `llvm-dis -version`. The version should be 3.5.0. 

Also, have you tried updating pip before installing the modified version of Tensorflow?
python -m pip install --upgrade pip

I installed LeFlow before in my machine (not the VM) and remember I had to deal with those kinds of problems, but it worked. If this ends up taking a lot of time I would encourage you to use the VM, since generating the circuits using LeFlow is generally very quick. After generating the design you can move the results out of the VM if needed.

jk78346, 2019-06-24:
> Hi Kuan-Chieh,
>
> I'm not sure, but it looks like LeFlow is trying to use a different version from LLVM than what it is supposed to. You can check this by running the command `llvm-dis -version`. The [...]

On my system it is:
```
LLVM (http://llvm.org/):
  LLVM version 3.8.0
  
  Optimized build.
  Built Jul  9 2016 (11:22:59).
  Default target: x86_64-pc-linux-gnu
  Host CPU: knl
```

danielholanda, 2019-06-26:
Hi Kuan-Chieh,

Try installing LLVM 3.5 instead and let me know if that works.

Also, I received your email and will reply to it as soon as I can.

closed 2019-09-18 by danielholanda