danielholanda/LeFlow #24: What is the relationship between the return value of legup and the return value of tensorflow?

cdyangzhenyu, 2019-09-17:
@danielholanda Hi danniel

I try leflow to trans tensorflow code to verilog, but I meet some question. 

My code:
```
import tensorflow as tf
import numpy as np
import sys
sys.path.append('../../src')
import processMif as mif

tf.logging.set_verbosity(tf.logging.INFO)
size=3

in_a = np.array([1,2,3])
in_b = np.array([0,1,2])
mif.createMem([in_a,in_b])

with tf.Session() as sess:
        x = tf.placeholder(tf.float32,[size])
        z = tf.placeholder(tf.float32,[size])
        with tf.device("device:XLA_CPU:0"):
            y=tf.nn.softmax(x*z)
        print in_a
        print in_b
        print y
        result = sess.run(y, {x: in_a, z: in_b})
        print(result)
```

The result is: [0.00242826 0.01794253 0.9796292 ]

But when I use ModelSim with `make v`, It return:

```
# Expect performance to be adversely affected.
# run 7000000000000000ns
# At t=             8470000 clk=1 finish=1 return_val=1051372203
# Cycles:                  421
# ** Note: $finish    : vecmul_max.v(5825)
#    Time: 8470 ns  Iteration: 2  Instance: /main_tb
# End time: 11:53:32 on Sep 17,2019, Elapsed time: 0:00:01
# Errors: 0, Warnings: 1
```
The return_val is 1051372203,  and it's float value is 0.3333333432674408

**My question is:**
- What is the relationship between the return value of legup and the return value of tensorflow?
- Tensorflow returns an array, but legup only returns one value. Why?
- How to use the legup return val ?

Looking forward to your answer, thank you

cdyangzhenyu, 2019-09-17:
I think I find some answer: #20 #21

cdyangzhenyu, 2019-09-17:
ok, I find the output ram from Modelsim. But what is the use of return_val?

[image]

danielholanda, 2019-09-18:
Hi,

Sorry for the delay. Glad you found the answer to most of your questions already.

return_val is only needed during compilation to avoid some computations from being optimized away. It has no use in the hardware itself. 

As you already figured out, the output is often a large matrix and is stored into an on-chip ram. If you look at the all rams generated you will notice that about half of them are called param* (e.g. param0, param1, param...). Those are the inputs, including weights and bias. There are also other memories called temp* (temp0, temp1, etc). Those are the activations. The last temp* memory is your output memory. 

Hope this helps

cdyangzhenyu, 2019-09-20:
Hi,

I'm glad to have your reply.

I have another question, How to confirm the relationship between param and input value, why x should correspond to param1 not param0, and what is the principle?

My code is as follows, If I change the order of param in `mif.createMem`, an error will be reported when `make v` is executed.

```
with tf.device("device:XLA_CPU:0"):
        hp1 = tf.nn.max_pool(tf.nn.relu(tf.add(tf.nn.conv2d(tf.reshape(x, [-1,28,28,1]), w_c1, strides=[1, 1, 1, 1], padding='SAME'),b_c1)),ksize=[1, 2, 2, 1], strides=[1, 2, 2, 1], padding='SAME')
        hp2 = tf.nn.max_pool(tf.nn.relu(tf.add(tf.nn.conv2d(hp1, w_c2, strides=[1, 1, 1, 1], padding='SAME'), b_c2)),ksize=[1, 2, 2, 1], strides=[1, 2, 2, 1], padding='SAME')
        h_fc1 = tf.nn.relu(tf.add(tf.nn.conv2d(hp2, w_fc1, strides=[1, 1, 1, 1], padding='VALID'), b_fc1))
        y = tf.nn.softmax(tf.add(tf.matmul(tf.reshape(h_fc1, [-1, 20]), w_fc2), b_fc2))

    ret = session.run(y, feed_dict={x: [mnist_data.test.images[test_image]]})

    print("Expected Result: "+str(np.argmax(mnist_data.test.labels[test_image])))
    print("Real Result: "+str(ret))

    # Creating memories for testing

    param1 = mnist_data.test.images[test_image]
    param0 = w_c1.eval()
    param2 = b_c1.eval()
    param3 = w_c2.eval()
    param4 = b_c2.eval()
    param5 = w_fc1.eval()
    param6 = b_fc1.eval()
    param7 = w_fc2.eval()
    param8 = b_fc2.eval()
    mif.createMem([param0,param1,param2,param3,param4,param5,param6,param7,param8])
```

danielholanda, 2019-09-20:
Good question.

The IR emitted by XLA doesn't come with any information that can be used to map the params to the Tensorflow inputs, so it is tricky to know what is the correct order that you should use in the createMem function. To create the mif files I recommend going to <your_file>_ir_7.ll and check the list of params (inputs) and their sizes. You can match things in this way.

Creating a way to dump this information from XLA and automatically using it when creating mif files would be a very nice contribution to LeFlow.

cdyangzhenyu, 2019-09-21:
Thank you for your reply. I'll try again.

closed 2019-09-21 by cdyangzhenyu