danielholanda/LeFlow #26: How to read memory_dump.txt

alewarne, 2019-10-17:
I successfully compiled and ran the classificationMNIST example. The memory_dump.txt looks like this:
// memory data file (do not edit the following line - required for mem load use)
// instance=/main_tb/top_inst/memory_controller_inst/temp3/ram
// format=hex addressradix=h dataradix=h version=1.0 wordsperline=1 noaddress
3a85f6b5
39abb140
3c6c273d
3a0045f5
3c3ac058
3b034bcc
3f73939e
3a0428d1
3b214716
3c82549b

What is this actually saying me?
Thank you in advance

danielholanda, 2019-10-17:
Hi,

This is the network output encoded in Hex. Those 10 outputs correspond to the confidence level of digits 0-9. You can use something very similar to this to convert it to float:
https://github.com/danielholanda/LeFlow/blob/master/examples/maxpoolMNIST/displayResult.py

alewarne, 2019-10-17:
Ah, thank you. Thus, each call to "sesssion.run(something)" creates lines in the memory_dump file?

In this case the line:
session.run(y,{x: [mnist_data.test.images[123]]}) 
?

danielholanda, 2019-10-17:
Not exactly.

First, LeFlow generates Verilog. This verilog has a testbench. The testbench contains one instruction that saves the final output to a file. Thus, this file is only generated when you run the testbench (normally using Modelsim). This means that the output that you get is the actual hardware output, not the Tensorflow software output.

alewarne, 2019-10-17:
Yes, I meant this. But then, how exactly do you specify (in source code) _which_ is the value that is written to the file? In the verilog testbench module it says

$writememh("memory_dump.txt",top_inst.memory_controller_inst.temp3.ram);

But why is temp3 chosen actually? It seems a bit arbitrary at this point?

danielholanda, 2019-10-17:
Let's say that this is our network:

    # Generating circuit
    with tf.device("device:XLA_CPU:0"):

        # Convolution 1
        conv1 = tf.nn.relu(tf.add(tf.nn.conv2d(X, conv1_w, strides=[1, 1, 1, 1], padding='SAME'),conv1_b))

        # Maxpool 1 
        maxp1=tf.layers.max_pooling2d(inputs=conv1, pool_size=[2, 2], strides=2)

        # Convolution 2
        conv2 = tf.add(tf.nn.conv2d(maxp1, conv2_w, strides=[1, 1, 1, 1], padding='SAME'),conv2_b)

        # Maxpool 2
        maxp2=tf.layers.max_pooling2d(inputs=conv2, pool_size=[2, 2], strides=2)

        # Reshaping array
        maxp2_reshaped = tf.nn.relu(tf.reshape(maxp2, shape=[1, 784]))

        # Dense layer 1 
        dense1 = tf.add(tf.matmul(maxp2_reshaped, dense1_w)[0], dense1_b)

        # Dense layer 2
        dense1_reshaped = tf.reshape(dense1, shape=[1, 20])
        y = tf.add(tf.matmul(dense1_reshaped, dense2_w)[0], dense2_b)

    sess.run(tf.global_variables_initializer())
    result = sess.run(y,{X: test_image})

Since "y" is the output of the network, this is what the testbench is going to dump.

I agree that the names in the verilog are hard to match with the names in the Tensorflow specification. Note that all arrays are mapped to on-chip memories (same that LegUp does). In the Verilog you will see multiple memories called param* (e.g. param1, param2, ...): those are the weights. You will also see temp* (e.g. temp1, temp2, ...): those are the activations and the final output. Usually, your last temp corresponds to the output of the network.

alewarne, 2019-10-18:
I see. Thank you a lot!

closed 2019-10-18 by danielholanda