danielholanda/LeFlow #27: Xilinx support

ChowderII, 2020-01-09:
Hello David!

Thank you for this amazing project! 
As I saw in your other issues, it seems complicated if not impossible to get your project to work with Xilinx devices. What exactly would one need to do to enable support for Xilinx. I have access to Vivado HLS (not the source code) so I would like to use that instead of LegUp if at all possible.

Thanks for your help!

danielholanda, 2020-01-13:
Hi ChowderII,

If you simply want to use LeFlow with Xilinx devices you could copy and paste the generic Verilog generated by LeFlow into a project that you manually created for a Xilinx board.
 
It is not possible to use LeFlow without LegUp, since this is the tool that we use to map the LLVM IR to Verilog. The only way to use Vivado HLS would be to find a way to feed it with LLVM IR instead of C and adapt LeFlow to work with this new tool. I'm pretty sure that Xilinx has an easy way to use IR instead of C, but this information is not public. 

Daniel

ChowderII, 2020-01-13:
Thank you very much for the quick reply!

I am not entirely certain what you mean by copy paste the verilog code to a
Xilinx project. Isn't the code generated by LegUp board specific? I assume
it instantiates memory components and the likes for the specific
architecture you choose during translation?

Thank you for your help!
ChowderII

danielholanda, 2020-01-13:
If I remember correctly the type of memory used is actually inferred by the synthesis tool. Some boards might have floating-point blocks and LegUp will generate Verilog to use those. If you choose a board that doesn't have those blocks, the generated Verilog should be fairly generic (but quite inefficient).

ChowderII, 2020-01-13:
Hmm I see, perhaps I could generate the same model in verilog for my board
and a supported board and see the differences? I'll look into it.

Thank you very much!

closed 2020-01-17 by ChowderII