danielholanda/LeFlow #30: I have everything installed but still i am getting this error while running "python test_all.py --fast"

Sandeep2017, 2020-02-22:
legup@legup-vm:~/LeFlow-master/test$ python test_all.py --fast

Running test for 03_vecmul_b_f
	Generating the circuit...
Traceback (most recent call last):
  File "../../src/LeFlow", line 200, in <module>
    run_leflow(args.file_name,tool_path)
  File "../../src/LeFlow", line 79, in run_leflow
    shutil.copy(project_folder+"ir/ir-cluster_0__XlaCompiledKernel_true__XlaNumConstantArgs_0__XlaNumResourceArgs_0__module-with-opt.ll",project_folder+project_name+"_ir_1.ll")
  File "/usr/lib/python2.7/shutil.py", line 119, in copy
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 82, in copyfile
    with open(src, 'rb') as fsrc:
IOError: [Errno 2] No such file or directory: '03_vecmul_b_f_files/ir/ir-cluster_0__XlaCompiledKernel_true__XlaNumConstantArgs_0__XlaNumResourceArgs_0__module-with-opt.ll'
	Generating new inputs and running Tensorflow with them...
	Testing circuit using Modelsim with new inputs...
cp: cannot stat ‘03_vecmul_b_f_files/tfArgs/param*.mif’: No such file or directory
/home/legup/legup-4.0/examples//../llvm/Release+Asserts/bin/llvm-dis: Could not open 03_vecmul_b_f.prelto.1.bc: No such file or directory

make[1]: *** [all] Error 1
make: *** [03_vecmul_b_f.v] Error 2
Traceback (most recent call last):
  File "test_all.py", line 92, in <module>
    modelsim_result = np.array(mif.getModelsimMem(folder+"_files/memory_dump.txt"))
  File "../src/processMif.py", line 77, in getModelsimMem
IOError: [Errno 2] No such file or directory: '03_vecmul_b_f_files/memory_dump.txt'

danielholanda, 2020-02-23:
Hi Sandeep,

This looks like a Tensorflow problem. I suspect that there is a problem with the current version of Tensorflow we are using. Try to uninstall the current version (pip uninstall tensorflow) and use [this version](https://github.com/danielholanda/LeFlow/blob/7504ea9a5fcde898d6962eca5efbc5602bb522f2/src/tensorflow/tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl) instead.

Please let me know if it works.

closed 2020-03-07 by danielholanda

houston988, 2020-04-01:
hi, I am facing this issue too. The installation of the modified version of Tensorflow was done until the step "sudo pip install tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl --ignone-installed six". (Python 2.7 reached the end of its life on Jan 2020 showed, not sure about it is the root cause?)

Do i need to proceed the "Compiling from scratch" step as mentioned in readme.md?

danielholanda, 2020-04-01:
Hi,
You should not need to compile it from scratch. Are you using the legup VM and using the old Tensorflow version I mentioned in my previous comment?

houston988, 2020-04-02:
Yes, im using VM with LegUp4.0, the modified version tensorflow installlation done below, some warnings showed:

legup@legup-vm:~/LeFlow/src/tensorflow$ sudo pip install tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl --ignore-installed six
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
WARNING: The directory '/home/legup/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Processing ./tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl
/usr/local/lib/python2.7/dist-packages/pip/_vendor/urllib3/util/ssl_.py:380: SNIMissingWarning: An HTTPS request has been made, but the SNI (Server Name Indication) extension to TLS is not available on this platform. This may cause the server to present an incorrect TLS certificate, which can cause validation failures. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
  SNIMissingWarning,
/usr/local/lib/python2.7/dist-packages/pip/_vendor/urllib3/util/ssl_.py:139: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
  InsecurePlatformWarning,
Collecting six
/usr/local/lib/python2.7/dist-packages/pip/_vendor/urllib3/util/ssl_.py:139: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
  InsecurePlatformWarning,
  Downloading six-1.14.0-py2.py3-none-any.whl (10 kB)
Collecting enum34>=1.1.6
  Downloading enum34-1.1.10-py2-none-any.whl (11 kB)
Collecting absl-py>=0.1.6
  Downloading absl-py-0.9.0.tar.gz (104 kB)
     |████████████████████████████████| 104 kB 2.9 MB/s 
Collecting tensorboard<1.7.0,>=1.6.0
  Downloading tensorboard-1.6.0-py2-none-any.whl (3.0 MB)
     |████████████████████████████████| 3.0 MB 8.7 MB/s 
Collecting gast>=0.2.0
  Downloading gast-0.3.3-py2.py3-none-any.whl (9.7 kB)
Collecting astor>=0.6.0
  Downloading astor-0.8.1-py2.py3-none-any.whl (27 kB)
Collecting wheel
  Downloading wheel-0.34.2-py2.py3-none-any.whl (26 kB)
Collecting termcolor>=1.1.0
  Downloading termcolor-1.1.0.tar.gz (3.9 kB)
Collecting grpcio>=1.8.6
  Downloading grpcio-1.28.0-cp27-cp27mu-manylinux2010_x86_64.whl (2.8 MB)
     |████████████████████████████████| 2.8 MB 144 kB/s 
Collecting protobuf>=3.4.0
  Downloading protobuf-3.11.3-cp27-cp27mu-manylinux1_x86_64.whl (1.3 MB)
     |████████████████████████████████| 1.3 MB 5.6 MB/s 
Collecting mock>=2.0.0
  Downloading mock-3.0.5-py2.py3-none-any.whl (25 kB)
Collecting backports.weakref>=1.0rc1
  Downloading backports.weakref-1.0.post1-py2.py3-none-any.whl (5.2 kB)
Collecting numpy>=1.13.3
  Downloading numpy-1.16.6-cp27-cp27mu-manylinux1_x86_64.whl (17.0 MB)
     |████████████████████████████████| 17.0 MB 3.9 MB/s 
Collecting bleach==1.5.0
  Downloading bleach-1.5.0-py2.py3-none-any.whl (17 kB)
Collecting werkzeug>=0.11.10
  Downloading Werkzeug-1.0.1-py2.py3-none-any.whl (298 kB)
     |████████████████████████████████| 298 kB 4.8 MB/s 
Collecting html5lib==0.9999999
  Downloading html5lib-0.9999999.tar.gz (889 kB)
     |████████████████████████████████| 889 kB 6.2 MB/s 
Collecting markdown>=2.6.8
  Downloading Markdown-3.1.1-py2.py3-none-any.whl (87 kB)
     |████████████████████████████████| 87 kB 9.4 MB/s 
Collecting futures>=3.1.1; python_version < "3"
  Downloading futures-3.3.0-py2-none-any.whl (16 kB)
Collecting setuptools
  Downloading setuptools-44.1.0-py2.py3-none-any.whl (583 kB)
     |████████████████████████████████| 583 kB 3.0 MB/s 
Collecting funcsigs>=1; python_version < "3.3"
  Downloading funcsigs-1.0.2-py2.py3-none-any.whl (17 kB)
Building wheels for collected packages: absl-py, termcolor, html5lib
  Building wheel for absl-py (setup.py) ... done
  Created wheel for absl-py: filename=absl_py-0.9.0-py2-none-any.whl size=121931 sha256=9a120ba0061f33207a1f5656c5047c527d2d970d1ac388dace409940e5ef23aa
  Stored in directory: /tmp/pip-ephem-wheel-cache-mCt5l4/wheels/37/83/b0/40d9e9f3d5a7021dfda2f3ea1f0088235679cd7747761c7b93
  Building wheel for termcolor (setup.py) ... done
  Created wheel for termcolor: filename=termcolor-1.1.0-py2-none-any.whl size=4831 sha256=94c3eceb0e1382cfa5dee01c89c34d5c502cd575ed7edc923e708e09f324c21a
  Stored in directory: /tmp/pip-ephem-wheel-cache-mCt5l4/wheels/48/54/87/2f4d1a48c87e43906477a3c93d9663c49ca092046d5a4b00b4
  Building wheel for html5lib (setup.py) ... done
  Created wheel for html5lib: filename=html5lib-0.9999999-py2-none-any.whl size=107220 sha256=8e1e6182a4310ba24fa4228dd63aec9c1cdffaa6db6e45098481e0b99b23688e
[... 10 lines cut ...]

Running test for 03_vecmul_b_f
[... cut ...]
/home/legup/legup-4.0/examples//../llvm/Release+Asserts/bin/llvm-dis: Could not open 03_vecmul_b_f.prelto.1.bc: No such file or directory

make[1]: *** [all] Error 1
make: *** [03_vecmul_b_f.v] Error 2
Traceback (most recent call last):
  File "test_all.py", line 92, in <module>
    modelsim_result = np.array(mif.getModelsimMem(folder+"_files/memory_dump.txt"))
  File "../src/processMif.py", line 77, in getModelsimMem
IOError: [Errno 2] No such file or directory: '03_vecmul_b_f_files/memory_dump.txt'

danielholanda, 2020-04-12:
I just downloaded a clean version of the legup VM and tried to install LeFlow again to make sure that everything still works. 

**This is exactly what I did:**
Downloaded the legup VM
Cloned the latest version of LeFlow and the old Tensorflow version I mentioned above
Opened the terminal and used those commands:
    sudo apt-get update
    sudo apt-get install python-pip
    sudo python -m pip install --upgrade pip
    sudo pip install tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl --ignore-installed six

After this, I'm able to run "python test_all.py --fast" and all tests were successful.

houston988, 2020-04-13:
may i know what is your LeFLow directory in VM?

danielholanda, 2020-04-13:
I cloned the LeFlow repo to the Desktop of the LegUp VM.

mu-jchung, 2021-03-17:
I just ran into this issue. The current version of src/tensorflow/tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl will install tensorflow-1.6.0 that will refuse to be imported on certain machines with this error:

"The TensorFlow library was compiled to use AVX2 instructions, but these aren't available on your machine."

The problem was fixed by the older version of the wheel that was linked to in danielholanda's February 22 comment.