danielholanda/LeFlow #31: ERROR: Legup failed to generate Verilog

phatberstromgg, 2020-03-17:
Hi @danielholanda, your project is great. I have complied success all your examples. But when I tried to build my owm simple CNN (I follow your instruction in the other answer), I met ERROR: Legup failed to generate Verilog.
This is my code

import tensorflow as tf
import numpy as np
import sys
import matplotlib.image as mpimg
sys.path.append('../../src')

import processMif as mif 

def rand(dim):
    return np.random.random(dim).astype(np.float32)

X = tf.placeholder(tf.float32, [1, 227, 227, 3])
img=mpimg.imread('test.png')
img_1 = np.resize(img, (227,227,3))
test_image = np.reshape(img_1,[1,227,227,3])

conv1_w = tf.Variable(rand([3, 3, 3, 32]))
conv1_b = tf.Variable(rand([32]))

conv2_w = tf.Variable(rand([3, 3, 32, 32]))
conv2_b = tf.Variable(rand([32]))

conv3_w = tf.Variable(rand([3, 3, 32, 64]))
conv3_b = tf.Variable(rand([64]))

fc1_w = tf.Variable(rand([28*28*64, 128]))
fc1_b = tf.Variable(tf.constant(0.05, shape=[128]))

fc2_w = tf.Variable(rand([128, 2]))
fc2_b = tf.Variable(tf.constant(0.05, shape=[2]))

with tf.Session() as sess:
    with tf.device("device:XLA_CPU:0"):

        #Conv1
        conv1 = tf.nn.relu(tf.add(tf.nn.conv2d(X, conv1_w, strides=[1,1,1,1], padding='SAME'), conv1_b))

        #Maxpool1
        maxp1 = tf.layers.max_pooling2d(inputs=conv1, pool_size=[2,2], strides=2)

        #Conv2
        conv2 = tf.nn.relu(tf.add(tf.nn.conv2d(maxp1, conv2_w, strides=[1,1,1,1], padding='SAME'), conv2_b)) 

        #Maxpool2
        maxp2 = tf.layers.max_pooling2d(inputs=conv2, pool_size=[2,2], strides=2)

        #Conv3
        conv3 = tf.nn.relu(tf.add(tf.nn.conv2d(maxp2, conv3_w, strides=[1,1,1,1], padding='SAME'), conv3_b))

        #Maxpool3
        maxp3 = tf.layers.max_pooling2d(inputs=conv3, pool_size=[2,2], strides=2)

        #Flatten
        maxp3_reshaped = maxp3.get_shape()
        num_features = maxp3_reshaped[1:4].num_elements()
        layers = tf.reshape(maxp3, [-1, num_features])
       

        #Fc1
        fc1 = tf.nn.relu(tf.matmul(layers, fc1_w) + fc1_b)

        #Fc2
        fc2 = tf.matmul(fc1, fc2_w) + fc2_b

        y_pred = tf.nn.softmax(fc2, name='y_pred')
        # y_pred_cls = tf.argmax(y_pred, dimension=1)

    sess.run(tf.global_variables_initializer())
    result = sess.run(y_pred,{X: test_image})

    print("Ket qua:"+str(result))

danielholanda, 2020-03-17:
Hi,

You can try to get started from this code here. I just tested and it works fine.

```python
import tensorflow as tf
import numpy as np
import sys
sys.path.append('../../src')

import processMif as mif

# This creates a random 32-bit floating point input for your variables
def rand(dim):
    return np.random.random(dim).astype(np.float32)

X = tf.placeholder(tf.float32, [1, 28,28,1])
test_image=rand([1,28,28,1])

conv1_w = tf.Variable(rand([5,5,1,8]))
conv1_b = tf.Variable(rand([8])) 

conv2_w = tf.Variable(rand([5,5,8,16])) 
conv2_b = tf.Variable(rand([16]))

dense1_w = tf.Variable(rand([784, 20])) 
dense1_b = tf.Variable(rand([20]))

dense2_w = tf.Variable(rand([20, 10])) 
dense2_b = tf.Variable(rand([10]))

with tf.Session() as sess:

    # Generating circuit
    with tf.device("device:XLA_CPU:0"):

        # Convolution 1
        conv1 = tf.nn.relu(tf.add(tf.nn.conv2d(X, conv1_w, strides=[1, 1, 1, 1], padding='SAME'),conv1_b))

        # Maxpool 1 
        maxp1=tf.layers.max_pooling2d(inputs=conv1, pool_size=[2, 2], strides=2)

        # Convolution 2
        conv2 = tf.add(tf.nn.conv2d(maxp1, conv2_w, strides=[1, 1, 1, 1], padding='SAME'),conv2_b)

        # Maxpool 2
        maxp2=tf.layers.max_pooling2d(inputs=conv2, pool_size=[2, 2], strides=2)

        # Reshaping array
        maxp2_reshaped = tf.nn.relu(tf.reshape(maxp2, shape=[1, 784]))

        # Dense layer 1 
        dense1 = tf.add(tf.matmul(maxp2_reshaped, dense1_w)[0], dense1_b)

        # Dense layer 2
        dense1_reshaped = tf.reshape(dense1, shape=[1, 20])
        y = tf.add(tf.matmul(dense1_reshaped, dense2_w)[0], dense2_b)

    sess.run(tf.global_variables_initializer())
    result = sess.run(y,{X: test_image})

    # Print expected result
    print("Result Calculated: "+str(result))

    # Creating .mif files to initialize the memories
    # To do this, go to <your_file>_ir_7.ll and check the list of params (inputs) and their sizes 
    param8 = dense2_b.eval();
    param7 = dense2_w.eval();
    param6 = dense1_b.eval();
    param5 = dense1_w.eval();
    param4 = conv2_b.eval();
    param3 = conv2_w.eval();
    param2 = conv1_b.eval();
    param1 = test_image;
    param0 = conv1_w.eval();
    mif.createMem([param0,param1,param2,param3,param4,param5,param6,param7,param8])
```

Also, make sure that you are using this slightly older version of Tensorflow:
https://github.com/danielholanda/LeFlow/blob/7504ea9a5fcde898d6962eca5efbc5602bb522f2/src/tensorflow/tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl

The latest version from our Tensorflow seems to not have been properly compiled. This was also the version that allowed Tensorflow to properly generate IR for larger circuits. If you need LeFlow to generate larger circuits, you might need to recompile it and see if it works. All files are here: https://github.com/danielholanda/LeFlow/pull/14

Hope this helps

phatberstromgg, 2020-03-31:
Hi @danielholanda, thank you for your advice. I have had successfully generated the verilog code from tensorflow. 
Could you share me the code for what you did in the demo video. I want to give the input an image like the way you do and the ouput show the result in hex display.

danielholanda, 2020-03-31:
Hi,

The code for the demo is definitely very messy and this is why I decided to not put it here, but if you want I can email that to you. Just send me an email and I will reply to you with the code attached.

phatberstromgg, 2020-04-05:
Hi @danielholanda, I am sorry for my late response. This is my email: phat.vo.110717@gmail.com
I really appreciate your help. Thank you.

danielholanda, 2020-04-06:
Done!

closed 2020-04-06 by danielholanda