danielholanda/LeFlow #33: Using Xilinx FPGAs and AXI memory interfaces

ivanfv, 2020-05-19:
Hello,

I am very interested on LeFlow. I am trying to use Xilinx FPGAs with AXI memory interfaces. As far as I know, LegUp supports to create this kind of memory interfaces by setting an argument. 

I was looking to LeFlow code and didn't find where to set up arguments for LegUp. Do you have any suggestion or direction where should I look for? Is it there any source file where all the arguments are set?

Thanks a lot,
Ivan

danielholanda, 2020-06-12:
Hi Ivan,

Sorry for the terrible delay in replying. 

I don't recall the open-source version of LegUp having AXI memory interface support. I suspect that this is only possible in their commercial version. Unfortunately, when using LeFlow, we are limited to directly reading and writing directly from/to on-chip memory.

ivanfv, 2020-06-15:
Hi Daniel,

I understand, thanks a lot for your reply.

Best,
Ivan

closed 2020-06-15 by ivanfv