danielholanda/LeFlow #35: Legup 4.0 virtual image is not accessible

LinasEin, 2020-10-25:
Hi,
I would like to convert Tensorflow code into HDL, however provided Legup virtual machine link does not seem to be working - (http://legup.eecg.utoronto.ca). Tried [this](http://legup.eecg.utoronto.ca/legup-4.0-ubuntu-64-14.04.vdi) one (from [#18](https://github.com/danielholanda/LeFlow/issues/18)) - no luck either.
Where else can I find Leflow supported Legup 4.0 version?
Thanks!

robotron5exe, 2020-11-13:
Hi,
same problem here. By using legup 8.0 and changing examples folder i am getting error when running python test script.
Error is that there is no specific example that is i guess present in legup 4.0.
Thanks for help!!

danielholanda, 2020-11-14:
Hi folks,

Legup recently got acquired by Microchip. Since then, they have removed their page with the free version of LegUp (4.0) from their website. We are trying to get in touch with them to get the VM back out there (since it is open source). I will keep this issue open until we solve it.

juannzou, 2020-12-17:
Hi @danielholanda,
Any updates? I am also facing the same issue. Can't get into using LeFlow in the meantime!
Looking forward to your support.
Thanks.

danielholanda, 2020-12-21:
I followed up with the LegUp people. Looks like they are just having some server issues that have been made worse because of the pandemic. The plan is to have the website back soon. In the meanwhile, if you need access to LeFlow urgently, feel free to email me directly.

Sowee0, 2021-09-01:
Unfortunately the image is still unavailable. I have also messaged you @danielholanda to no avail. Is the email in your profile active?

danielholanda, 2021-09-01:
Yes, feel free to email me.

monkstein88, 2024-08-15:
Is the offer still valid, can I also get a link to the image? 
Thanks!

Duckie2369, 2025-03-20:
@danielholanda Is the image still available, sir? I have emailed you about the problem for a while and I want to know if the project is still able to work with. Thanks!