danielholanda/LeFlow #36: Cannot run tests

habibagamal, 2021-01-03:
I'm using Ubuntu 18.04.4 LTS
During the installation, I downloaded the code, then did the following commands
```
$ sudo apt-get install python-pip
$ sudo python -m pip install --upgrade pip
$ sudo pip install tensorflow
$ sudo pip install numpy
$ cd LeFlow-master/src/
$ chmod +x LeFlow
$ cd ../test/
$ python test_all.py --fast
```
This fails at the first test which is 03_vecmul_b_f. Also note that  `leflow_output` returned from the subprocess in test_all.py is `('', None)`

Find below a dump of the output

```
Running test for 03_vecmul_b_f
	Generating the circuit...
	Generating new inputs and running Tensorflow with them...
	Testing circuit using Modelsim with new inputs...
cp: cannot stat '03_vecmul_b_f_files/tfArgs/param*.mif': No such file or directory
make: *** No rule to make target 'v'.  Stop.
Traceback (most recent call last):
  File "test_all.py", line 91, in <module>
    modelsim_result = np.array(mif.getModelsimMem(folder+"_files/memory_dump.txt"))
  File "../src/processMif.py", line 77, in getModelsimMem
IOError: [Errno 2] No such file or directory: '03_vecmul_b_f_files/memory_dump.txt'
```

What may be causing this? Thanks

danielholanda, 2021-01-06:
Looks like Legup is not being found or has not been properly installed. have you tested LegUp separately? Also, if you are not using the VM you have to change the "legup_examples_folder" variable in the main LeFlow file.

habibagamal, 2021-01-07:
Do I need to do anything during the installation other than the steps I specified in my previous post? 
How can I test LegUp separately? 
I noticed the examples directory you're referring to in the LeFlow file, why do I need the examples folder when running the test_all.py script that should be running the python files in the test directory instead of examples directory? Should I change this path to the path of examples directory or test directory on my machine? 

Thanks.

danielholanda, 2021-01-18:
Looks like a good soul uploaded Legup 4.0 documentation here:
https://manualzz.com/doc/7507428/legup-documentation---university-of-toronto

This should hopefully help you test Legup separately.

For your second question: You should change that path to the path that Legup is installed (more specifically, the examples folder). This is needed to avoid that you need to do everything from the LegUp folder.

closed 2021-04-09 by danielholanda