danielholanda/LeFlow #39: fpga performance

soans1994, 2021-09-08:
hello author,

1. i wanted to ask about the performance of the fpga in real time camera.  Is it fast enough to be used in self driving cars?
2. can the tensorflow keras models be converted using leflow? also the models must be trained according to the fpga camera resolution?

sorry for silly  questions.
thank you

danielholanda, 2021-09-08:
1 - LeFlow is definitely not fast enough for being used in real-time self-driving car applications. 
2 - To map Tensorflow scripts to Verilog you need to follow the syntax shown in the examples directory. Ideally, you should train your model and perform inference using the same resolution.

closed 2021-09-08 by danielholanda

soans1994, 2021-09-21:
> 1 - LeFlow is definitely not fast enough for being used in real-time self-driving car applications.
> 2 - To map Tensorflow scripts to Verilog you need to follow the syntax shown in the examples directory. Ideally, you should train your model and perform inference using the same resolution.

thank you for your reply. I want to ask about the scope of deploying cnn models in fpga. Is it widely used? what is the advantage of using fpga's. i had learnt about fpgas in my graduation courses, but i dont know how it is used in practical applications. I want to check my cnn models performance on hardware. I was thinking about nvidia jetson nano/tx2 boards. Are FPGAs more powerful?

thank you

danielholanda, 2021-09-22:
FPGAs might offer a better performance-per-watt and better latency (but often not better thoughput depending on the workload). I recommend reading about project Brainwave (from Microsoft) as a start. If you simply just want to make something work, GPUs are easier to start with.