danielholanda/LeFlow #42: Unsupported Wheel Issue

jayrmh, 2022-05-24:
Hello, how can I solve the issue of `No distributions at all found for pip in /usr/lib/python2.7/dist-packages Storing debug log for failure in /home/legup/.pip/pip.log` on LegUp as well as```
 tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl is not a supported wheel on this platform.
Storing debug log for failure in /home/legup/.pip/pip.log
```

All instructions on the front page have been followed, but does not work. 
Kindly assist, thank you!

danielholanda, 2022-06-06:
Sorry for the delay. I'm not sure if this is your problem, but hopefully, this will help.

Looks like some computers have problems installing the last whl file, but the one before that seems to work fine for all architectures.

https://github.com/danielholanda/LeFlow/blob/7504ea9a5fcde898d6962eca5efbc5602bb522f2/src/tensorflow/tensorflow-1.6.0-cp27-cp27mu-linux_x86_64.whl

Sumaya10, 2022-08-09:
Hi!
Was this issue resolved? I'm also getting same issue and have installed whl file whose link is provided but still getting same issue.